mobilise-d/mobgap #120: GSD error metrics throw if reference did not detect a GS

AKuederle, 2024-03-26:
At the moment we get a divide by 0 error when the reference did not detect a GS. However, we should return NaN for the metrics that can not be calculated and not throw, as it could be valid that no gait happened in a recording.

AKuederle, 2024-03-26:
Maybe a similar approach as with the f1-Score calculation would make sense. Allow the user if it should warn, raise or be ignored

AKuederle, 2024-04-02:
I also realised, that the settings for what should happen with "zero-divide" are not exposed in the higher level functions. She should probably add this :)

a-mosquito, 2024-04-10:
I added a `zero_division_hint` parameter to the `calculate_unmatched_gsd_performance_metrics` function. I named it differently than the parameter used for the basic evaluation functions as it behaves differently
- set to `False`: warning is suppressed and values are set to nan
- set to "warn": warning is raised and values are set to nan
- set to "raise": `ZeroDivisionError` is raised

(comp. to `zero_division` parameter from f1-score etc: result is either set to 0, 1, or warning is raised)

closed 2024-04-17 by AKuederle