mobilise-d/mobgap #128: Find consistent naming for the DummyPipelines

AKuederle, 2024-04-15:
For each algo group we are creating a DummyPipeline that uses refrence info to mock the previous algo steps and has a score method that compares the performance against the reference.

This can be used during evaluation (hence, naming it something related might be good). However, it can also be used during training of the algorithm (e.g. for the ML algo). So it might make sense to have naming that makes that clear

AKuederle, 2024-04-16:
ChatGpt suggestion: GsdEmulatorPipeline -> As in emulating the rest of the pipeline

pltsc18, 2024-04-16:
Some ideas: 

- GsdSimulationPipeline: This name suggests that it simulates or mimics the behavior of the preceding algorithms using reference data. 
- GsdProxyPipeline: This name suggests that the pipeline acts as a proxy for the preceding algorithms, providing a stand-in representation for evaluation or training.
- GsdReplicaPipeline: "Replica" implies that the pipeline replicates the behavior of the preceding algorithms based on reference data.
- GsdMirrorPipeline: This name suggests that the pipeline mirrors the behavior of the preceding algorithms, reflecting their actions for evaluation or training purposes.
- GsdEmulatorPipeline: "Emulator" implies that the pipeline emulates the behavior of the preceding algorithms, mimicking their actions using reference data.
- GsdEchoPipeline: "Echo" suggests that the pipeline echoes the behavior of the preceding algorithms, reproducing their actions for evaluation or training.