mobilise-d/mobgap #134: Suggestions to rename top-level folders to make them easier to undertand for "outstiders"

AKuederle, 2024-05-03:
At the moment we use shorthands for most algorithms (GSD, SL, ...). While these might be easy to be understood within the Mobilise-D consortium, they might be confusing.

I would suggest to at least change the top-level folders of the these algorithm groups to better to understand names.

Suggestions:

gsd -> gait_sequences
cad -> cadence
icd -> inital_contacts
td -> turning
lrc -> laterality/left_right? <- Not sure about that one yet
wba -> maybe group under the aggregation folder?

Note, that I suggest skipping the implied "detection" in the names as well. I think this should be clear and helps to keep the names somewhat short.

The algorithms itself still carry the abbreviation in the name. I think this is a good approach and we should keep it.

pltsc18, 2024-05-03:
I totally agree, I also like "laterality" for `lrc` :)

alexstihi, 2024-05-10:
I agree, this new naming convention is much more clearer to external users. For L/R detection, how about `contact_side`?

DMegaritis, 2024-05-16:
Fully agree as well! 
For lrc I would suggest "left_right" since "laterality" implies the choice of using one side of body by humans.

PR #156 (2024-06-17): Big rename
Closes #134
commit 6f3bade9 (2024-06-17): Big rename (#156)

Closes #134
closed 2024-06-17 by AKuederle