mobilise-d/mobgap #146: Add citation file

pltsc18, 2024-06-07:
The guide describes how to properly cite the package depending on its use. It is also indicated to use, when appropriate, the link to mobgap. In this regard, GitHub offers the option of attaching a CITATION.cff file to the repository that allows you to cite the repo with a simple copy and paste. It might be useful to add this feature! https://academia.stackexchange.com/questions/14010/how-do-you-cite-a-github-repository

AKuederle, 2024-10-30:
@pltsc18 Could you actually create a citation file for the Repo? Ask Andrea who to include. I would suggest for the repo citation only people that actually worked on it + WP2/3 leads. More people can be included in official paper

pltsc18, 2024-10-30:
On it!

pltsc18, 2024-10-30:
CFF citation files allow to include various pieces of information about the work. I would be grateful if you could give me some feedback regarding some of these:
- **persistent identifiers**: In the CFF we can add persistent identifiers such as DOIs, Software Heritage numbers, or ArXiv URLs. Do we have anything like that?
- **Related resources**: the CFF file allows to indicate URLs for code repository, website/landing page, and other repository. I was thinking at this association: _code repository_ --> https://github.com/mobilise-d/mobgap/ ; _website_ --> https://mobgap.readthedocs.io/en/latest/README.html ; _other repository_ --> https://github.com/mobilise-d/Mobilise-D-TVS-Recommended-Algorithms
- **Abstract**: I though at a short description to help possible users know at a glance if this work is what they are looking for "A Python implementation of the Mobilise-D algorithm pipeline for gait analysis using IMU worn at the lower back (Learn more about [the Mobilise-D project](https://mobilise-d.eu/)). This package is meant as reference implementation for research and production use."
- **Keywords**: What about _gait analysis,  walking speed estimation, wearable sensors, end-to-end pipeline_, and _ mobilise-d_?

AKuederle, 2024-10-31:
identifier: Not yet. Will ask Dave and Brian to make a Zenodo release for the Code. Then we have a DOI (we can update the version there later)
Related Recources: Don't need other rescources I think.
Abstract: Sounds good
Keywords: Also sounds good (don't think this does much)

pltsc18, 2024-11-05:
Thanks for your feedback. The citation file has been committed, let me know if that could work!

AKuederle, 2024-11-05:
Great! Can you also update the "recommeded citation" in the README to suggest using the new information?

pltsc18, 2024-11-05:
Done, let me know if it that can work!

closed 2024-11-05 by AKuederle