mobilise-d/mobgap #152: Add example for evaluation DMOs that are already aggregated (e.g., per day)

a-mosquito, 2024-06-10:
**Please consider also using the Discussion tab for general questions or ideas, in particular if they are not linked to the development, but rather scientific questions/discussions.**

**Describe the issue**
A clear and concise description of what the issue is.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

a-mosquito, 2024-06-12:
change naming from gait sequences to WB in text and variables, explain how both evaluation approaches from WBA paper would be implemented

AKuederle, 2024-06-13:
I already adapted the naming and some of the structure. What is missing, is the part of the example that shows the "aggregate then error" approach. For this we should probably move the detailed explanations on how the error calculation works up into the new part (or have a completely seperate section that explain the details?). Whatever you think is best to keep everything readable

closed 2024-06-28 by AKuederle