mobilise-d/mobgap #198: multi_groupby can have unexpected results when input dfs are empty

AKuederle, 2025-01-27:
The fact that the output structure is wrong, when no groups exists might be a general problem. Can you open a dedicated issue for that? Unclear at the moment what the best solution is. But maybe raise an error to force the user to handle the case explictly... Let's discuss in new issue

_Originally posted by @AKuederle in https://github.com/mobilise-d/mobgap/pull/196#discussion_r1890545930_

metinbicer, 2025-01-27:
This occurs when the first argument (`primary_df`) is empty.

AKuederle, 2025-04-11:
Implemented more of a workaround instead of a fix. As there is no universal solution to determine the expected outputshape in case the primary df is empty, we raise an explicit descriptive error now, that the user can handle. This way it is clear what is happening, however, the user still has to implement the logic on their own.

commit 9516e55e (2025-04-11): Implemented explicit handling of for cases where primary df is empty in a multi-df groupby. Closes #198
closed 2025-04-11 by AKuederle