mobilise-d/mobgap #203: Failing cadence algorithms for the laboratory data

metinbicer, 2025-02-04:
The new algos (`IcdShinImproved` and `IcdHKLeeImproved`) fail on a couple more WBs than the old ones.

| algo          | version | count |
| ------------- | ------- | ----- |
| HKLeeImproved | new     | 114.0 |
|               | old     | 113.0 |
| ShinImproved  | new     | 119.0 |
|               | old     | 113.0 |

The extra failing trials for the new algos are:

| cohort | participant_id | test_name_pretty     | algo          | reference_ws |
|--------|----------------|---------------------|---------------|--------------|
| PFF    | 4064           | Straight Walk Slow  | HKLeeImproved | 0.267943     |
| CHF    | 2079           | Straight Walk Slow  | ShinImproved  | 0.407844     |
| MS     | 2025           | Straight Walk Slow  | ShinImproved  | 0.519717     |
| MS     | 2027           | Straight Walk Slow  | ShinImproved  | 0.429368     |
| MS     | 3033           | Straight Walk Slow  | ShinImproved  | 0.531800     |
| MS     | 4040           | Straight Walk Slow  | ShinImproved  | 0.459100     |
| PD     | 3008           | Straight Walk Slow  | ShinImproved  | 0.338373     |

This should be further investigated.

AKuederle, 2025-02-04:
We should investigate, what failing means here? Is this a "planned fail", where some condition of the algorithm is not met or an "accidental fail", why we throw an error, or devide by 0 or something like this

AKuederle, 2025-02-06:
On a similar note, I found a couple of SL Wbs that had NaN as walking speed. We should also look into this.

AKuederle, 2025-04-11:
@metinbicer Is this worth investigating further? If yes, could you look into this?

metinbicer, 2025-04-11:
It doesn't seem to be a planned fail. `stride_list_with_approx_paras` becomes all NaN due to the first index of `cadence_per_sec` being NaN. This happens due to not interpolating the first index as explained below. Specifically for the following datapoints:
```
TVSLabDatasetGroupLabel(cohort='PFF', participant_id='4064', time_measure='TimeMeasure1', test='Test6', trial='Trial1', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
TVSLabDatasetGroupLabel(cohort='CHF', participant_id='2079', time_measure='TimeMeasure1', test='Test6', trial='Trial2', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
TVSLabDatasetGroupLabel(cohort='MS', participant_id='2025', time_measure='TimeMeasure1', test='Test6', trial='Trial2', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
TVSLabDatasetGroupLabel(cohort='MS', participant_id='2027', time_measure='TimeMeasure1', test='Test6', trial='Trial1', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
TVSLabDatasetGroupLabel(cohort='MS', participant_id='3033', time_measure='TimeMeasure1', test='Test6', trial='Trial2', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
TVSLabDatasetGroupLabel(cohort='MS', participant_id='4040', time_measure='TimeMeasure1', test='Test6', trial='Trial1', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
TVSLabDatasetGroupLabel(cohort='PD', participant_id='3008', time_measure='TimeMeasure1', test='Test6', trial='Trial2', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')
```

The main reason is that [`mobgap.utils.interpolation.interval_mean`](https://github.com/mobilise-d/mobgap/blob/main/mobgap/utils/interpolation.py#L14) returns NaN in the first index. This can be reproduced for the followings (values are for the datapoint `TVSLabDatasetGroupLabel(cohort='CHF', participant_id='2079', time_measure='TimeMeasure1', test='Test6', trial='Trial2', test_name='WalkSlow', test_name_pretty='Straight Walk Slow')` ):

```
import numpy as np
from mobgap.utils.interpolation import interval_mean
measurements = np.array([ 1.13,  2.08,  3.  ,  3.83,  4.78,  5.62,  6.94,  8.06,  9.02, 10.1 ])
measurement_samples = np.array([0.95, 0.92, 0.83, 0.95, 0.84, 1.32, 1.12, 0.96, 1.08, 1.08])
interval_start_ends = np.array([[ 0.,  1.],
       [ 1.,  2.],
       [ 2.,  3.],
       [ 3.,  4.],
       [ 4.,  5.],
       [ 5.,  6.],
       [ 6.,  7.],
       [ 7.,  8.],
       [ 8.,  9.],
       [ 9., 10.],
       [10., 11.]])
interval_mean(measurements, measurement_samples, interval_start_ends)
# array([  nan, 0.95 , 0.875, 0.89 , 0.84 , 1.32 , 1.12 ,   nan, 0.96 , 1.08 , 1.08 ])
# Note that not only the first value but also an intermediate one is missing
```

After this is returned in `robust_step_para_to_sec` function as `step_time_per_sec`, it gets smoothed and masked to interpolate missing values:
https://github.com/mobilise-d/mobgap/blob/b5d3d05bd5aada5db4dc5246294ce5ab5125e832/mobgap/utils/interpolation.py#L119-L126

As it says above, it interpolates within the recording. The second missing value above is interpolated as 1.1 (between 1.12 and 1.08). However, the first index stays missing and it is returned to `cadence_per_sec_`. Not fully sure if this would be something we want. I guess when the first initial contact happens later than 1 sec, we will have this issue due to `interval_mean`:
https://github.com/mobilise-d/mobgap/blob/b5d3d05bd5aada5db4dc5246294ce5ab5125e832/mobgap/utils/interpolation.py#L51

AKuederle, 2025-04-11:
Thanks for looking into this! A couple of initial thoughts:

- This is a unfortunate conceptional issue on how we use the cadence algorithms. In the pipeline, they are applied to a refined gs. A refined GS is already cut so that the start and the end value must be an IC. The CAD algorithm (which internally also depends on some form of IC detection, then searches for suitable ICs, and will almost definitely not identify the start and the end as  IC, as it does not see enough information around it to find it... There are ways around that. We could add the start and the end of the GS manually to the IC list here (https://github.com/mobilise-d/mobgap/blob/b5d3d05bd5aada5db4dc5246294ce5ab5125e832/mobgap/cadence/_cad_from_ic.py#L308). I also thought about providing the GS with some padding around CAD and SL algo, but this is quite complicated with the structure we have now (namely the gs-iteration stuff). This would also be a major deviation from the original pipeline.
- So if we have a single NaN we get NaN for the full GS? We might want to change the mean across the strides to nan mean then... Seems to be better for these types of edgecases

metinbicer, 2025-04-14:
> * So if we have a single NaN we get NaN for the full GS? We might want to change the mean across the strides to nan mean then... Seems to be better for these types of edgecases

Isn't it already nan mean?

https://github.com/mobilise-d/mobgap/blob/b5d3d05bd5aada5db4dc5246294ce5ab5125e832/mobgap/utils/interpolation.py#L58-L60

Instead, I think we should skip when no ic is within any sec_center. So, if we drop missing ones `interval_means = interval_means[~np.isnan(interval_means )]` after the above mean calculation,the rest should be fine.

AKuederle, 2025-04-15:
Ah yes you are right! Btw. I checked if adding the start and the end of the GS as a forced IC is a good idea... It is not... It makes the first and the last step extremely short (at least when using the reference ICs to define the GSs)

This also indicates that that the CAD-IC algos find quite different ICs then the reference system...

AKuederle, 2026-09-28:
Decided on a solution: We now keep seconds that don't contain steps explicitly as NaN, so that we can still recover the data from the rest of the GS.

Cadence can be `NaN` near the start or end of a gait sequence, or in an internal gap longer than `max_interpolation_gap_s`. Short internal gaps are interpolated as before. Previously, one remaining `NaN` could make every later stride’s cadence `NaN` as well. The fix keeps strides that overlap the missing interval as `NaN` while preserving valid results for later strides.

closed 2026-09-28 by AKuederle