mobilise-d/mobgap #214: Applying constant rotation to a dataset

rmndrs89, 2025-03-14:
https://github.com/mobilise-d/mobgap/blob/c9235faa7e67fb52a4ea89e3bd310be09e2d6168/mobgap/_gaitmap/utils/rotations.py#L40

In order to align my IMU data with the coordinate systems defined by mobgap [here](https://mobgap.readthedocs.io/en/stable/guides/coordinate_system.html), I need to simply rotate my data 180 degrees around the x-axis. My data would then be aligned with the sensor frame.

I try to do this with:

```python
from mobgap.utils.rotations import rotate_dataset_series
from scipy.spatial.transform import Rotation

raw_data = load_imu_data(file_path)
sf_data = rotate_dataset_series(dataset=raw_data, rotations=Rotation.from_rotvec([np.pi, 0., 0.]))
```

which throws an error:

> Exception has occurred: TypeError
Single rotation has no len().
  File "D:\Projects\mobgap\mobgap\_gaitmap\utils\rotations.py", line 40, in rotate_dataset_series
    if len(dataset) != len(rotations):
  File "D:\Projects\mobgap\main.py", line 30, in main
    sf_data = rotate_dataset_series(
  File "D:\Projects\mobgap\main.py", line 39, in <module>
    main()
TypeError: Single rotation has no len().

I see that you want as many Rotation objects as there are data pionts. Would you then supply it as a list, or array, like
```python
rotations = [
    Rotation.from_rotvec([np.pi, 0, 0])
    for _ in range(len(raw_data))
]
```

AKuederle, 2025-03-14:
Good point. The function for that usecase does not exist at the moment.
Will add that later.

You can just call `_rotate_sensor` directly with the single rotation that you have.

As a general note, all the rotations are based on scipy rotation objects. Each rotation object can either represent a single rotation or a series of rotations. When applying it to data numpy broadcasting rules apply (see here form more detail: https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.apply.html).

AKuederle, 2025-03-14:
Btw. if that is to slow, you can also take the "flip_dataset" function from gaitmap

closed 2026-09-25 by AKuederle